TinyTapeout/tinytapeout_www #66: Homepage - ambiguous Tiles/PCBs count

DeflateAwning, 2024-02-28:
Can you please confirm if the counts on the homepage (e.g., "21/100 tiles") are the number of remaining tiles or the number of sold tiles?

It would be awesome if it said "21/100 tiles sold" or "21/100 tiles available" to make it more clear.

urish, 2024-03-11:
@mattvenn any idea how to address it without cluttering the page?

mattvenn, 2024-03-11:
make the 21 the same colour as the bar chart? 

put a 'sold so far' above the charts?

append 'sold' to each stat: '21/100 tiles sold' ?

DeflateAwning, 2024-03-11:
I'm a fan of appending the word "sold"

urish, 2024-03-11:
thoughts?

[image]

DeflateAwning, 2024-03-11:
Looks good to me! Alternative, if you like it more:
[image]

urish, 2024-03-12:
How about just showing percentages instead of numbers?

[image]

mattvenn, 2024-03-12:
yes, even simpler.

urish, 2024-03-12:
There we have it. Thanks again @DeflateAwning for the feedback!

closed 2024-03-12 by urish