TinyTapeout/tinytapeout_www #99: Minor updates on 'Analog pins' page

algofoogle, 2024-08-01:
On the https://tinytapeout.com/specs/analog/ page:

1. [Price of $380](https://github.com/TinyTapeout/tinytapeout_www/blob/47716111f635dd5da2a08d02320e496dfe6dc05c/content/specs/analog/_index.md?plain=1#L59) (and subsequent breakdown) should be $320 according to: 2x2 tiles @ $50ea = $200, 3 analog pins @ $40ea = $120. If the first 4 analog pins are $40ea, and others are $100ea, a better example might be: 2x2 tiles + **5** analog pins = $200 + $100 + $160 = $460.
2. I think the word [owl](https://github.com/TinyTapeout/tinytapeout_www/blob/47716111f635dd5da2a08d02320e496dfe6dc05c/content/specs/analog/_index.md?plain=1#L113) should be "layout"?
3. The max current *per pin* is [specified as 4 mA](https://github.com/TinyTapeout/tinytapeout_www/blob/47716111f635dd5da2a08d02320e496dfe6dc05c/content/specs/analog/_index.md?plain=1#L23) but is there a max current spec as a whole for the [4 different templates](https://github.com/TinyTapeout/tinytapeout_www/blob/47716111f635dd5da2a08d02320e496dfe6dc05c/content/specs/analog/_index.md?plain=1#L82-L87) (i.e. max current for the power gating)?

urish, 2024-08-01:
Thanks for sharing your feedback!

> [Price of $380](https://github.com/TinyTapeout/tinytapeout_www/blob/47716111f635dd5da2a08d02320e496dfe6dc05c/content/specs/analog/_index.md?plain=1#L59) (and subsequent breakdown) should be $320 [...]

Actually, the example talks about a second design on the same shuttle, so the first two pins are still discounted (at $40 / pin), and then the third pin is full price ($100). Any suggestions on

 how to rephrase the example so it's more clear that it talks about two designs on the same shuttle?

> I think the word [owl](https://github.com/TinyTapeout/tinytapeout_www/blob/47716111f635dd5da2a08d02320e496dfe6dc05c/content/specs/analog/_index.md?plain=1#L113) should be "layout"?

@mattvenn wrote it - I believe it's a reference to the following meme:

[image]

> max current for the power gating?

@smunaut any input on this one?

> [Spanish version](https://github.com/TinyTapeout/tinytapeout_www/blob/main/content/specs/analog/_index.es.md) still refers to 8 analog pins instead of 6

@mattvenn who's in charge of keeping the translation up to date? Can you please tag them?

algofoogle, 2024-08-01:
Thanks @urish for explaining! Your clarification on price makes sense; it's basically counted per the user's use of analog pins on the shuttle, not per project. Maybe the replacement for this line, and the one above it, could be:
> In addition, there is a cost for each analog pin attributable to a given customer (rather than project) on a given shuttle. The price for analog pins is $40 per pin for the first 4 pins you need on the Tiny Tapeout shuttle, and $100 per pin for additional pins.
>
> Example:
> * You submit a design requiring 1x2 tiles and 2 analog pins. This costs you $180 ($100 for the tiles, and $80 for the analog pins).
> * You then submit another design on the same shuttle, itself requiring 2x2 tiles and 3 analog pins. This would cost you $380 ($200 for the tiles, and $180 for the analog pins).
> * i.e. as a customer you are using a total of **5** analog pins on the shuttle.

Haha, I must've taken a vacation from the Internet and missed that meme. Makes sense now.

smunaut, 2024-08-01:
We recommend to stay under 10 mA for single height tiles and 20 mA for dual tile height.
We'll have more precise number when we get TT06 back since it has some PDN caracterization tests.

It's also not so much a "max" current, because you could probably pull more without doing any kind of damage, it's just the impedance of the PDN causes voltage drop and at some point it become unreasonable.

algofoogle, 2025-02-04:
Do we have some indication now of "rated" current per project, per different tile counts?
> 3. The max current per pin is [specified as 4 mA](https://github.com/TinyTapeout/tinytapeout_www/blob/47716111f635dd5da2a08d02320e496dfe6dc05c/content/specs/analog/_index.md?plain=1#L23) but is [...]

smunaut, 2025-02-04:
At room temperature, the max rating would be ~ 50 mA per tile height. ( so 50 mA for h=1, 100 mA for h=2, 200 mA for h=4 ).
But that's the rating of when the wire would break down.

But the guidelines above are still applicable, they correspond to ~ 0.1V drop on VDD. For a h=2 project the data we have is ~ 4.5 ohm VDPWR impedance and ~ 1 ohm VGND impedance.

We don't have data yet for the 3.3V power gate that will only come whenever we get TT8 back.