TinyTapeout/tinytapeout_www #107: h3 is bigger than h2

mattvenn, 2024-09-10:
make h3 smaller than h2.

urish, 2024-09-14:
The original sizes of H1/H2 (coming from the hugo theme we're using) are:
```css
h1 {
  font-size: 3.25rem; 
}

h2 {
  font-size: 2.5rem;
  line-height: 110% !important;
  margin: 2.5rem 0 1.5rem 0;
}
```

However, we have our own tinytapeout-theme.css file that overrides them to be smaller:

```css
h1 {
  font-size: 1.5em;
}

h2 {
  font-size: 1.3em;
}
```

**This is what things look like right now (with our theme overrides):**

[image]

**This is what they'd look like if we removed the override, and used the default sizes:**

[image]

**Should I remove our overrides ?**

mattvenn, 2024-09-15:
yeah remove them I think

urish, 2024-09-15:
Done

closed 2024-09-15 by urish