TinyTapeout/tinytapeout_www #119: Show Analog project pinout using A0-5 B0-5 notation

smunaut, 2024-11-05:
Ideally both the raw index (as is now) and the A0-5 / B0-5 notation would be shows. Former is useful when tracing the actual track in the SPEF to see parasitics for instance. The latter is what most people will use to wire stuff up.

urish, 2024-11-05:
How about this? 

[image]

Any changes (especially to the terms) or does it look good?

urish, 2024-11-05:
Also, double checking that I got the logic right, does this look ok?: 
```ts
  const breakoutPin = (analog: number) => {
    return analog < 6 ? 'A' + analog : 'B' + (analog - 6);
  };
```

smunaut, 2024-11-05:
Yes, that looks right to me