TinyTapeout/tinytapeout_www #131: Include more-complete pinouts in Tech Specs section

algofoogle, 2025-02-04:
Several times I've had a reason to check out something to do with the typical TT QFN-64 pinout, expecting to find it in the [Tech Specs](https://tinytapeout.com/specs/) section. A pin table would be sufficient. An added pin diagram would be nifty! By the way, if it helps, there is [this SVG & table](https://algofoogle-caravel-docs.readthedocs.io/en/latest/pinout.html) (as yet unreleased) as a possible template.

It would also be nice to have the same thing for the carrier (breakout) board(s).

mattvenn, 2025-02-05:
Don't we have this in the datasheet?

[image]

[image]

you're saying we should put this also in the tech spec section?

urish, 2025-02-07:
It may makes sense to include them in the shuttle pages, since they change between shuttles (tt04/05 vs tt06 through tt10 vs ttihp0p2)